modbus2mqtt/modbus2mqtt #13: unable to edit specification

albgit, 2024-09-06:
Hi,
I've installed this addon and beginning to configure my heatpump connected through a Wifi Modbus RTU to TCP adapter (USR W610).
As a test I've configured a specification (named hmi-120) with a single entity and I see that in MQTT Explorer.
The problem is that when I try to change the specification (with the pencil button) it opens a blank specification with no data.
I was especting to continue and add others entity, but I can't.
I couldn't even find the file. 

-----------------------------------------------------------

 Add-on: Modbus <=> MQTT
 Service converts RTU and TCP Modbus to MQTT from other
-----------------------------------------------------------
 Add-on version: 0.16.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.1  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.8.2
 Home Assistant Supervisor: 2024.08.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
9/6/2024 8:30:16 PM notice modbus2mqtt module dir: /usr/app/node_modules/@modbus2mqtt/angular/dist/browser/en-US/index.html
9/6/2024 8:30:16 PM notice modbus2mqtt http root : /usr/app/node_modules/@modbus2mqtt/angular/dist/browser
From https://github.com/modbus2mqtt/modbus2mqtt.config
   3f8be01..c7506a5  main       -> origin/main
9/6/2024 8:30:17 PM notice bus Updating 3f8be01..c7506a5
9/6/2024 8:30:17 PM notice bus Fast-forward
9/6/2024 8:30:17 PM notice bus  .github/workflows/validate.yml | 21 +++++++++++++++------
9/6/2024 8:30:17 PM notice bus  1 file changed, 15 insertions(+), 6 deletions(-)
9/6/2024 8:30:17 PM notice bus 
9/6/2024 8:30:18 PM notice config File not found: /data/public/specifications/files/eastronsdm720-m/files.yaml
9/6/2024 8:30:18 PM notice config specifications directory not found /data/contributed/specifications
9/6/2024 8:30:18 PM notice HttpServerBase Hassio authentication successful
9/6/2024 8:30:18 PM notice modbus2mqtt modbus2mqtt listening on  37e92509-modbus2mqtt: 3000
9/6/2024 8:35:13 PM notice modbuscache readDiscreteInputs TIMEOUT: slave:3 address: 3 length:1  retrying ... 
9/6/2024 8:35:13 PM notice modbuscache readInputRegisters TIMEOUT: slave:3 address: 0 length:76  retrying ... 
9/6/2024 8:35:52 PM ERR! m2mSpecification Error: No options available for entity id: -1
9/6/2024 8:35:52 PM ERR! m2mSpecification     at SelectConverter.getOptions (/usr/app/node_modules/@modbus2mqtt/specification/dist/selectConverter.js:31:15)
9/6/2024 8:35:52 PM ERR! m2mSpecification     at SelectConverter.modbus2mqtt (/usr/app/node_modules/@modbus2mqtt/specification/dist/selectConverter.js:46:32)
9/6/2024 8:35:52 PM ERR! m2mSpecification     at M2mSpecification.copyModbusDataToEntity (/usr/app/node_modules/@modbus2mqtt/specification/dist/m2mspecification.js:438:50)
9/6/2024 8:35:52 PM ERR! m2mSpecification     at M2mSpecification.fileToModbusSpecification (/usr/app/node_modules/@modbus2mqtt/specification/dist/m2mspecification.js:394:43)
9/6/2024 8:35:52 PM ERR! m2mSpecification     at Modbus.populateEntitiesForSpecification (/usr/app/node_modules/@modbus2mqtt/server/dist/modbus.js:96:54)
9/6/2024 8:35:52 PM ERR! m2mSpecification     at /usr/app/node_modules/@modbus2mqtt/server/dist/modbus.js:110:20
9/6/2024 8:35:52 PM ERR! m2mSpecification  Error: No options available for entity id: -1
9/6/2024 8:35:52 PM ERR! m2mSpecification     at SelectConverter.getOptions (/usr/app/node_modules/@modbus2mqtt/specification/dist/selectConverter.js:31:15)
9/6/2024 8:35:52 PM ERR! m2mSpecification     at SelectConverter.modbus2mqtt (/usr/app/node_modules/@modbus2mqtt/specification/dist/selectConverter.js:46:32)
9/6/2024 8:35:52 PM ERR! m2mSpecification     at M2mSpecification.copyModbusDataToEntity (/usr/app/node_modules/@modbus2mqtt/specification/dist/m2mspecification.js:438:50)
9/6/2024 8:35:52 PM ERR! m2mSpecification     at M2mSpecification.fileToModbusSpecification (/usr/app/node_modules/@modbus2mqtt/specification/dist/m2mspecification.js:394:43)
9/6/2024 8:35:52 PM ERR! m2mSpecification     at Modbus.populateEntitiesForSpecification (/usr/app/node_modules/@modbus2mqtt/server/dist/modbus.js:96:54)
9/6/2024 8:35:52 PM ERR! m2mSpecification     at /usr/app/node_modules/@modbus2mqtt/server/dist/modbus.js:110:20
9/6/2024 8:36:57 PM ERR! config Unable to read Files directory for /data/local/specifications/files/hmi-120/files.yaml
9/6/2024 8:36:57 PM ERR! config {"errno":-2,"syscall":"open","code":"ENOENT","path":"/data/local/specifications/files/hmi-120/files.yaml"}
9/6/2024 8:37:15 PM notice config File not found: /data/local/specifications/files/hmi-120/files.yaml
9/6/2024 8:40:46 PM notice modbuscache readHoldingRegisters TIMEOUT: slave:3 address: 2 length:1  retrying ... 
9/6/2024 8:40:46 PM notice modbuscache Retry successfully executed: slave:3 address: 2 length:1
9/6/2024 8:41:47 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Data length error, expected 7 got 6
9/6/2024 8:51:47 PM notice modbuscache readHoldingRegisters TIMEOUT: slave:3 address: 2 length:1  retrying ... 
9/6/2024 8:51:47 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Data length error, expected 7 got 13
9/6/2024 9:01:46 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Data length error, expected 7 got 6
9/6/2024 9:07:02 PM notice modbuscache readHoldingRegisters TIMEOUT: slave:3 address: 2 length:1  retrying ... 
9/6/2024 9:07:02 PM notice modbuscache Retry successfully executed: slave:3 address: 2 length:1
9/6/2024 9:17:02 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Data length error, expected 7 got 10
9/6/2024 9:31:46 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Data length error, expected 7 got 4
9/6/2024 9:42:47 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Data length error, expected 7 got 14
9/6/2024 9:52:47 PM notice modbuscache readHoldingRegisters TIMEOUT: slave:3 address: 2 length:1  retrying ... 
9/6/2024 9:52:47 PM notice modbuscache Retry successfully executed: slave:3 address: 2 length:1
9/6/2024 10:02:46 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Data length error, expected 7 got 4
9/6/2024 10:08:32 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Data length error, expected 7 got 6
9/6/2024 10:15:19 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Data length error, expected 7 got 5
9/6/2024 10:34:17 PM notice modbuscache readHoldingRegisters TIMEOUT: slave:3 address: 2 length:1  retrying ... 
9/6/2024 10:34:17 PM notice modbuscache Retry successfully executed: slave:3 address: 2 length:1
9/6/2024 10:39:32 PM notice modbuscache Retry successfully executed: slave:3 address: 2 length:1
9/6/2024 10:43:46 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Data length error, expected 7 got 3
9/6/2024 10:51:46 PM notice modbuscache readHoldingRegisters slave: 3 startAddress: 2 length: 1: Unexpected data error, expected address 3 got 52

volkmarnissen, 2024-09-07:
Hi, 

You need to save the specification before closing the specifications page.
You can save it once all error messages are gone.
In the screem shot I see, that the documentation and the image is not configured.
Did you save the specification?
Otherwise, it is lost.
Did you get a warning when you left the specificatiion page?

So, it's a good idea,  to upload the image and the documentation first. This eliminates the error messages.
Then you add one entity and then you can save the specification.

Next week, there will be an update of the Addon and hopefully a new video about advanced usage.
But it won't cover this issue.

Volkmar

volkmarnissen, 2024-09-07:
Last week, I found a bug.
If you entered URL's for image and documentation, and then saved it. This information was not stored in the specification.

This will be solved next week with the update of the addon. As a workaround, you can download the image and the documentation and upload it to the addon and then save it. This will work with the current version.

Best Regards,

Volkmar

albgit, 2024-09-07:
I think my problem is related to the bug.
I saved the specification with all the informations, doc and image too ( otherwise the save button is grayed out and can't be saved).
I'll wait for the update and test it.
Can you say where the addon save its files?
I want to erase all the specifications that I've created.
Thanks
Regards

closed 2024-09-16 by volkmarnissen