modbus2mqtt/modbus2mqtt #17: Unable to change modbus poll interval

albgit, 2024-09-27:
I'm experiencing a lot of problems when I want to change my specification.
When I open my working specification to edit it freezes.
I think it's related to modbus poll interval .
The first is fixed to 1000 ms that may cause the timeouts in the log below (maybe 60000 ms can be better).
I tried to change it, but it won't go after 1000 ms.
The TCP connection timeout is related to the moment when I opened my specification.

 Add-on: Modbus <=> MQTT
 Service converts RTU and TCP Modbus to MQTT from other
-----------------------------------------------------------
 Add-on version: 0.16.7
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.1  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.9.2
 Home Assistant Supervisor: 2024.09.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
9/27/2024 7:37:47 PM notice HttpServerBase Hassio authentication successful
9/27/2024 7:37:47 PM notice modbus2mqtt modbus2mqtt listening on  37e92509-modbus2mqtt: 3000
9/27/2024 7:37:48 PM notice mqttdiscover New MQTT Connection 
9/27/2024 7:37:48 PM notice mqttdiscover poll: MQTT client reconnected
9/27/2024 7:37:48 PM notice mqttdiscover MQTT Connection will be closed by Last will
9/27/2024 7:37:49 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 0 length:45  retrying ... 
9/27/2024 7:37:49 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 0 length: 45: Data length error, expected 95 got 90
9/27/2024 7:37:49 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 117 length:27  retrying ... 
9/27/2024 7:37:51 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 29 length: 1: Data length error, expected 7 got 4
9/27/2024 7:38:02 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 0 length:45  retrying ... 
9/27/2024 7:38:02 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 0 length: 45: Data length error, expected 95 got 89
9/27/2024 7:38:02 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 117 length:27  retrying ... 
9/27/2024 7:38:03 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 16 length: 1: Data length error, expected 7 got 6
9/27/2024 7:38:16 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 0 length:45  retrying ... 
9/27/2024 7:38:16 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 0 length: 45: Data length error, expected 95 got 93
9/27/2024 7:38:16 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 117 length:27  retrying ... 
9/27/2024 7:38:17 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 10 length:1  retrying ... 
9/27/2024 7:38:17 PM notice modbuscache Retry successfully executed: slave:3 address: 10 length:1
9/27/2024 7:38:30 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 0 length:45  retrying ... 
9/27/2024 7:38:30 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 0 length: 45: Data length error, expected 95 got 93
9/27/2024 7:38:30 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 117 length:27  retrying ... 
9/27/2024 7:38:33 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 42 length: 1: Data length error, expected 7 got 4
9/27/2024 7:38:44 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 0 length:45  retrying ... 
9/27/2024 7:38:45 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 0 length: 45: Data length error, expected 95 got 88
9/27/2024 7:38:45 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 117 length:27  retrying ... 
9/27/2024 7:38:48 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 43 length:1  retrying ... 
9/27/2024 7:38:48 PM notice modbuscache Retry successfully executed: slave:3 address: 43 length:1
9/27/2024 7:38:57 PM ERR! bus poll(3) connection failed Error: TCP Connection Timed Out
9/27/2024 7:38:57 PM notice modbuscache poll(3) Retry Connect: module : TCP Connection Timed Out Error: TCP Connection Timed Out
9/27/2024 7:38:57 PM notice modbuscache     at Socket.<anonymous> (/usr/app/node_modules/modbus-serial/ports/tcpport.js:158:28)
9/27/2024 7:38:57 PM notice modbuscache     at Socket.emit (node:events:517:28)
9/27/2024 7:38:57 PM notice modbuscache     at Socket.emit (node:domain:489:12)
9/27/2024 7:38:57 PM notice modbuscache     at Socket._onTimeout (node:net:598:8)
9/27/2024 7:38:57 PM notice modbuscache     at listOnTimeout (node:internal/timers:569:17)
9/27/2024 7:38:57 PM notice modbuscache     at process.processTimers (node:internal/timers:512:7)
9/27/2024 7:38:57 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 0 length:45  retrying ... 
9/27/2024 7:38:57 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 0 length: 45: Data length error, expected 95 got 87
9/27/2024 7:38:57 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 117 length:27  retrying ... 
9/27/2024 7:38:58 PM notice modbuscache http(3) readHoldingRegisters TIMEOUT: slave:3 address: 0 length:45  retrying ... 
9/27/2024 7:38:58 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 5 length: 1: Data length error, expected 7 got 87
9/27/2024 7:38:58 PM notice modbuscache http(3) readHoldingRegisters TIMEOUT: slave:3 address: 117 length:27  retrying ... 
9/27/2024 7:38:58 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 6 length: 1: Data length error, expected 7 got 59
[... 12 lines cut ...]
9/27/2024 7:39:14 PM notice modbuscache     at Socket._onTimeout (node:net:598:8)
9/27/2024 7:39:14 PM notice modbuscache     at listOnTimeout (node:internal/timers:569:17)
9/27/2024 7:39:14 PM notice modbuscache     at process.processTimers (node:internal/timers:512:7)
9/27/2024 7:39:14 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 0 length:45  retrying ... 
9/27/2024 7:39:14 PM notice modbuscache http(3) readHoldingRegisters slave: 3 startAddress: 38 length: 1: Data length error, expected 7 got 87
9/27/2024 7:39:14 PM ERR! modbus Modbus Read (2,3) failed: Data length error, expected 7 got 87
9/27/2024 7:39:17 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 42 length:1  retrying ... 
9/27/2024 7:39:17 PM notice modbuscache Retry successfully executed: slave:3 address: 42 length:1
9/27/2024 7:39:27 PM ERR! bus poll(3) connection failed Error: TCP Connection Timed Out
9/27/2024 7:39:27 PM notice modbuscache poll(3) Retry Connect: module : TCP Connection Timed Out Error: TCP Connection Timed Out
9/27/2024 7:39:27 PM notice modbuscache     at Socket.<anonymous> (/usr/app/node_modules/modbus-serial/ports/tcpport.js:158:28)
9/27/2024 7:39:27 PM notice modbuscache     at Socket.emit (node:events:517:28)
9/27/2024 7:39:27 PM notice modbuscache     at Socket.emit (node:domain:489:12)
9/27/2024 7:39:27 PM notice modbuscache     at Socket._onTimeout (node:net:598:8)
9/27/2024 7:39:27 PM notice modbuscache     at listOnTimeout (node:internal/timers:569:17)
9/27/2024 7:39:27 PM notice modbuscache     at process.processTimers (node:internal/timers:512:7)
9/27/2024 7:39:27 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 0 length:45  retrying ... 
9/27/2024 7:39:27 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 0 length: 45: Data length error, expected 95 got 87
9/27/2024 7:39:27 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 117 length:27  retrying ... 
[... cut ...]
9/27/2024 7:39:41 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 117 length:27  retrying ... 
9/27/2024 7:39:55 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 0 length:45  retrying ... 
9/27/2024 7:39:55 PM notice modbuscache poll(3) readHoldingRegisters slave: 3 startAddress: 0 length: 45: Data length error, expected 95 got 94
9/27/2024 7:39:55 PM notice modbuscache poll(3) readHoldingRegisters TIMEOUT: slave:3 address: 117 length:27  retrying ...

volkmarnissen, 2024-09-28:
I assume, the polling parameter is not the reason for the timeouts.
However, I will check if it's really not possible to increase it more than 1000ms.

My timeout setting for the RTU devices at 9600 baud is 500ms. 

Please check the timeout parameter in the **"List of Modbus Controllers"** in the
menu **"Show busses ..."**
Set the parameter **"Timeout"** to **500ms** or 1000ms. 
If that doesn't help, restarting the app might help (Renewing the TCP/IP connection).
Alternatively, the reason can be at the slave side. If possible restart the Modbus Controller of the slave.

You could you use nodered addon to check if you are able to get the modbus data.
Nodered uses the same Modbus API as modbus2mqtt.
This API has an issue with Modbus timeouts, but modbus2mqtt uses a workaround for it.

If anything helped, please close the issue.
Leave a comment with your solution. That may help other users and helps me  to start a troubleshooting chapter in the documentation.

Otherwise, please send me the content of the directory
```
/usr/hassio/addons/*_modbus2mqtt
```
You can remove the local/secrects.yaml file before sending it.
It should be possible to reproduce it with this data.

The addon will be enhanced. It gets a download link for that. That should be available by mid of October.

albgit, 2024-09-29:
I connect my home assistant to a usr w610 that translate Modbus rtu to modbus tcp.
In modbus2mqtt I inserted the tcp connection (not rtu) to the w610 with a timeout of 100 (I tried to increase that, but I didin't work).
In the usr w610 the rtu connection is set with a baud rate of 9600 and configuring the modbus side of ha in configuration.yaml everything goes smooth (but I want to switch to modbus2mqtt, as I think is more flexible).
I also would like to say that in MQTTExplorer I have some modbus2mqtt output, but not every specification voice is showing. 
I want to attach the content of the directory, but i didn't found it (see terminal image below).

tcp config:
[image]

MQTT explorer:
[image]

Folder not found:
[image]

volkmarnissen, 2024-09-29:
100ms is definitly too short.
If that can't be changed, this is definitly a bug.
I'll look at it. However, a fix can only be available mid of next week.
I don't know, where the data directory for the addon in your installation is.
So, let's wait for the next release. It will have a link to download the configuration.

albgit, 2024-09-29:
I tried to increase the number, but without success.
I'll wait for the next releases.
If I can suggest It would be great to have, besides the link to download, also a text that show where the data directory is.

PR #18 (2024-10-16): New Features and fix for tcp/ip timeout
Feature: 
- Trigger Poll MQTT. In device configuration, you can now specify how a device can be polled. 
a) By fixed configurable interval (default)
b) by Trigger from MQTT
c) both
- Show MQTT topics for a given device:
In the devices menu, you can now view the topics used by a specific device.
In particular
a) the topic for the state of all entities of the device
b) the command topic, if the device is not read only
c) The trigger topic, if the device is configured to be triggered manually.

- Upload and download of Specifications
a) In the Specification [... cut]
closed 2024-10-16 by volkmarnissen