modbus2mqtt/modbus2mqtt #20: Version 0.16.8 bus poll error

albgit, 2024-10-18:
Hi,
I've just installed 0.16.8 version and in log appears a bus poll error.
I think It's linked to TCP timeout parameter that I set initially to 1000 ms and after to 2000 ms.

Here the log:

-----------------------------------------------------------
 Add-on: Modbus <=> MQTT
 Service converts RTU and TCP Modbus to MQTT from other
-----------------------------------------------------------
 Add-on version: 0.16.8
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.1  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.10.2
 Home Assistant Supervisor: 2024.10.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
10/18/2024 8:29:20 AM notice HttpServerBase Hassio authentication successful
10/18/2024 8:29:20 AM notice modbus2mqtt modbus2mqtt listening on  37e92509-modbus2mqtt: 3000
10/18/2024 8:29:20 AM notice mqttdiscover New MQTT Connection 
10/18/2024 8:29:20 AM notice mqttdiscover poll: MQTT client reconnected
10/18/2024 8:29:20 AM notice mqttdiscover MQTT Connection will be closed by Last will
10/18/2024 8:29:39 AM ERR! bus poll(3) connection failed TypeError [ERR_INVALID_ARG_TYPE]: The "msecs" argument must be of type number. Received type string ('1000')
10/18/2024 8:29:39 AM notice modbuscache poll(3) Retry Connect: module : The "msecs" argument must be of type number. Received type string ('1000') TypeError [ERR_INVALID_ARG_TYPE]: The "msecs" argument must be of type number. Received type string ('1000')
10/18/2024 8:29:39 AM notice modbuscache     at new NodeError (node:internal/errors:405:5)
10/18/2024 8:29:39 AM notice modbuscache     at validateNumber (node:internal/validators:177:11)
10/18/2024 8:29:39 AM notice modbuscache     at getTimerDuration (node:internal/timers:394:3)
10/18/2024 8:29:39 AM notice modbuscache     at Socket.setStreamTimeout [as setTimeout] (node:internal/stream_base_commons:244:11)
10/18/2024 8:29:39 AM notice modbuscache     at new TcpPort (/usr/app/node_modules/modbus-serial/ports/tcpport.js:96:43)
10/18/2024 8:29:39 AM notice modbuscache     at cl.connectTCP (/usr/app/node_modules/modbus-serial/apis/connection.js:118:22)
10/18/2024 8:29:39 AM notice modbuscache     at /usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:169:35
10/18/2024 8:29:39 AM notice modbuscache     at new Promise (<anonymous>)
10/18/2024 8:29:39 AM notice modbuscache     at Bus.connectRTUClient (/usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:153:16)
10/18/2024 8:29:39 AM notice modbuscache     at /usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:224:22
10/18/2024 8:29:39 AM ERR! bus poll(3) connection failed TypeError [ERR_INVALID_ARG_TYPE]: The "msecs" argument must be of type number. Received type string ('1000')
10/18/2024 8:29:39 AM notice modbuscache poll(3) Retry Connect: module : The "msecs" argument must be of type number. Received type string ('1000') TypeError [ERR_INVALID_ARG_TYPE]: The "msecs" argument must be of type number. Received type string ('1000')
10/18/2024 8:29:39 AM notice modbuscache     at new NodeError (node:internal/errors:405:5)
10/18/2024 8:29:39 AM notice modbuscache     at validateNumber (node:internal/validators:177:11)
10/18/2024 8:29:39 AM notice modbuscache     at getTimerDuration (node:internal/timers:394:3)
10/18/2024 8:29:39 AM notice modbuscache     at Socket.setStreamTimeout [as setTimeout] (node:internal/stream_base_commons:244:11)
10/18/2024 8:29:39 AM notice modbuscache     at new TcpPort (/usr/app/node_modules/modbus-serial/ports/tcpport.js:96:43)
10/18/2024 8:29:39 AM notice modbuscache     at cl.connectTCP (/usr/app/node_modules/modbus-serial/apis/connection.js:118:22)
10/18/2024 8:29:39 AM notice modbuscache     at /usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:169:35
10/18/2024 8:29:39 AM notice modbuscache     at new Promise (<anonymous>)
10/18/2024 8:29:39 AM notice modbuscache     at Bus.connectRTUClient (/usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:153:16)
10/18/2024 8:29:39 AM notice modbuscache     at /usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:185:22
10/18/2024 8:29:39 AM ERR! modbus Modbus Read (0,3) failed: The "msecs" argument must be of type number. Received type string ('1000')
10/18/2024 8:29:58 AM ERR! bus poll(3) connection failed TypeError [ERR_INVALID_ARG_TYPE]: The "msecs" argument must be of type number. Received type string ('1000')
10/18/2024 8:29:58 AM notice modbuscache poll(3) Retry Connect: module : The "msecs" argument must be of type number. Received type string ('1000') TypeError [ERR_INVALID_ARG_TYPE]: The "msecs" argument must be of type number. Received type string ('1000')
10/18/2024 8:29:58 AM notice modbuscache     at new NodeError (node:internal/errors:405:5)
10/18/2024 8:29:58 AM notice modbuscache     at validateNumber (node:internal/validators:177:11)
10/18/2024 8:29:58 AM notice modbuscache     at getTimerDuration (node:internal/timers:394:3)
10/18/2024 8:29:58 AM notice modbuscache     at Socket.setStreamTimeout [as setTimeout] (node:internal/stream_base_commons:244:11)
10/18/2024 8:29:58 AM notice modbuscache     at new TcpPort (/usr/app/node_modules/modbus-serial/ports/tcpport.js:96:43)
10/18/2024 8:29:58 AM notice modbuscache     at cl.connectTCP (/usr/app/node_modules/modbus-serial/apis/connection.js:118:22)
10/18/2024 8:29:58 AM notice modbuscache     at /usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:169:35
10/18/2024 8:29:58 AM notice modbuscache     at new Promise (<anonymous>)
10/18/2024 8:29:58 AM notice modbuscache     at Bus.connectRTUClient (/usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:153:16)
10/18/2024 8:29:58 AM notice modbuscache     at /usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:224:22
[... 64 lines cut ...]
10/18/2024 8:43:52 AM ERR! bus poll(3) connection failed TypeError [ERR_INVALID_ARG_TYPE]: The "msecs" argument must be of type number. Received type string ('2000')
10/18/2024 8:43:52 AM notice modbuscache poll(3) Retry Connect: module : The "msecs" argument must be of type number. Received type string ('2000') TypeError [ERR_INVALID_ARG_TYPE]: The "msecs" argument must be of type number. Received type string ('2000')
10/18/2024 8:43:52 AM notice modbuscache     at new NodeError (node:internal/errors:405:5)
10/18/2024 8:43:52 AM notice modbuscache     at validateNumber (node:internal/validators:177:11)
10/18/2024 8:43:52 AM notice modbuscache     at getTimerDuration (node:internal/timers:394:3)
10/18/2024 8:43:52 AM notice modbuscache     at Socket.setStreamTimeout [as setTimeout] (node:internal/stream_base_commons:244:11)
10/18/2024 8:43:52 AM notice modbuscache     at new TcpPort (/usr/app/node_modules/modbus-serial/ports/tcpport.js:96:43)
10/18/2024 8:43:52 AM notice modbuscache     at cl.connectTCP (/usr/app/node_modules/modbus-serial/apis/connection.js:118:22)
10/18/2024 8:43:52 AM notice modbuscache     at /usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:169:35
10/18/2024 8:43:52 AM notice modbuscache     at new Promise (<anonymous>)
10/18/2024 8:43:52 AM notice modbuscache     at Bus.connectRTUClient (/usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:153:16)
10/18/2024 8:43:52 AM notice modbuscache     at /usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:224:22
10/18/2024 8:43:52 AM ERR! bus poll(3) connection failed TypeError [ERR_INVALID_ARG_TYPE]: The "msecs" argument must be of type number. Received type string ('2000')
10/18/2024 8:43:52 AM notice modbuscache poll(3) Retry Connect: module : The "msecs" argument must be of type number. Received type string ('2000') TypeError [ERR_INVALID_ARG_TYPE]: The "msecs" argument must be of type number. Received type string ('2000')
10/18/2024 8:43:52 AM notice modbuscache     at new NodeError (node:internal/errors:405:5)
[... cut ...]
10/18/2024 8:43:52 AM notice modbuscache     at /usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:169:35
10/18/2024 8:43:52 AM notice modbuscache     at new Promise (<anonymous>)
10/18/2024 8:43:52 AM notice modbuscache     at Bus.connectRTUClient (/usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:153:16)
10/18/2024 8:43:52 AM notice modbuscache     at /usr/app/node_modules/@modbus2mqtt/server/dist/bus.js:185:22
10/18/2024 8:43:52 AM ERR! modbus Modbus Read (0,3) failed: The "msecs" argument must be of type number. Received type string ('2000')

volkmarnissen, 2024-10-21:
Can you please goto the menu ``` "contribute/import/export specifications"``` and ```"Download Local Settings"```.
Then attach the downloaded .zip file to this issue?
I have no tcp modbus controller. So it's difficult to reproduce it.

Thank you, Volkmar

volkmarnissen, 2024-10-21:
I was able to reproduce it w/o the configuration. There is no need to send the zip file anymore.