modbus2mqtt/modbus2mqtt #32: [BUG] TCP-Connection not kept alive

madejackson, 2024-10-28:
Seems like it's not keeping the TCP-Connection to the Modbus-TCP-Server alive. This will overload the network on fast polling rates.
[image]

modbus2mqtt, 2024-10-28:
Yes, that's true. However, I assume the typical poll interval is longer. May be, it makes sense to change the connection mode per configuration.

As of now, I'd like to leave it as it is. RTU connection uses the same approach. In that scenario, it's beneficial to close the connection.

madejackson, 2024-10-29:
> Yes, that's true. However, I assume the typical poll interval is longer. May be, it makes sense to change the connection mode per configuration.

Thanks for your response.

Unfortunately without supporting faster polling frequencies, you exclude any real time application like lights or shutter control. Shutter Control needs to be below 150ms to be usable, usually between 50-100ms.

volkmarnissen, 2024-11-11:
I get your point. We can leave the issue open.
It should also be possible to restrict closing the connection only when it's really needed.

This is not a small change. So, this won't be fixed in the next bug fixing releases.

arturmietek, 2025-02-03:
Today I also stared to using TCP gateway for modbus (now testing only one device) but my connection least only for about 1 hour. 
`2/3/2025 5:31:47 PM notice mqttdiscover Publish Discovery: length:3
2/3/2025 6:11:12 PM notice modbuscache poll(1) readDiscreteInputs TIMEOUT: slave:1 address: 0 length:1  retrying ... 
2/3/2025 6:15:00 PM notice modbuscache poll(1) readInputRegisters TIMEOUT: slave:1 address: 0 length:2  retrying ... 
2/3/2025 6:15:00 PM notice modbuscache Retry successfully executed: slave:1 address: 0 length:2
2/3/2025 6:25:22 PM ERR! modbuscache ModbusClient is open after close
2/3/2025 6:25:22 PM notice modbuscache poll(1) readInputRegisters slave: 1 startAddress: 0 length: 2: Port Not Open`

I use default pool rate 1000ms.