modbus2mqtt/modbus2mqtt #34: [BUG] Measurements missing "state_class: measurement"

madejackson, 2024-10-29:
Sorry for posting so much issues. Don't get me wrong, I like the work you have done and I just want to give feedback :)

Measurements in Home Assistant should be state_class: measurement to support [long-term-statistics](https://developers.home-assistant.io/docs/core/entity/sensor/?_highlight=state_class#long-term-statistics).

See a sensor I included twice, once directly via modbus and once via modbus2mqtt. You can see the better display on the first image as it uses min, max and median values to display.

[image]
[image]

modbus2mqtt, 2024-10-29:
I am very happy about your feedback. I will look into it in two weeks. It’s always interesting to get an idea from another viewpoints to the same thing. Please don’t stop sending your feedback.👍There is on thing I don’t understand. Why don’t you use home assistant supervisor?I installed it on Ubuntu, on odroid platform with Debian without any issue. It’s much easier than standalone HA.(which I used in my first approach)Only windows might be a bit difficult.There is a nice installation instruction and you a disable the OS check( for Ubuntu).Von meinem iPhone gesendet

image.png (view on web)
image.png (view on web)

madejackson, 2024-10-29:
There are several reasons I am against using HAOS. Just because it's simpler doesn't really cut it for me. The only real selling point of HAOS is Addons.  I don't like this limitation as it's an artificial design choice of HA for their own needs.

The same applies for OIDC support BTW. It's pretty clear by that point that HA does not want to add these features in for their own agenda ignoring the community. (Sources: [OIDC](https://community.home-assistant.io/t/open-letter-for-improving-home-assistants-authentication-system-oidc-sso/494223), [Addon's](https://community.home-assistant.io/t/addons-for-docker-installation/436190/85))

Some of my personal reasons against HAOS:
1. Home Assistant is an App which I am running on my own infrastructure which hosts 109 Apps right now. I don't want to run a dedicated VM for Home Assistant just because HA feels entitled to take over my OS. ANY other app runs perfectly fine without any limitation:
[image]

2. I am actively working with Cosmos Server, which is a new and very elegant approach for offering users a secure, easy to manage, independent Operating System. HAOS cannot do that while still taking over your machine.

volkmarnissen, 2024-11-11:
The state_class will be available in next release

closed 2024-11-15 by volkmarnissen