modbus2mqtt/modbus2mqtt #36: [FEAT] support for special values in Number registers

madejackson, 2024-10-30:
Sometimes, Modbus-Register use specific values for invalid or unavailable data (f.e. 0 or 65535). Would be awesome to be able to configure this. If modbus2mqtt then would read the configured value (f.e. 65535), it disables the data-point or sets it to invalid/unavailable.

madejackson, 2024-11-11:
You're talking about HA-Datatypes right? I think every value can have states different from a valid value.
This may or may not be represented either as a specific value or on a seperate register/coil to read out.

So this FR is more or less connected to #35

volkmarnissen, 2024-11-12:
OK.
For #36: If modbus has no value for a single entity, no value should be published for that entity. Other entities should still be published. Currently, all entites are published whether they have a modbus value or not.

For this specific topic, I see the following option.
A javascript addon for a specification could be uploaded like the documentation or the image file.
This addon can be assigned to the entity and preprocess and/or postprocess the data conversion from modbus to mqtt.
This is no small change and has to wait for a bigger release.
If an addon is used in multiple specifications, it can be included into modbus2mqtt and is then available for all specification s without uploading.

arturmietek, 2024-11-13:
Maybe I'll add my examples for SofarSolar inverters for consideration.

For one register the device status is defined as follows:
```
  lambda: |-
      auto z = "Unknown";
      char d = data[item->offset+1];
      if (d == 0) z = "Wait";
      else if (d == 1) z = "Check";
      else if (d == 2) z = "Normal";
      else if (d == 3) z = "Fault";
      else if (d == 4) z = "Permanent";
      return {z};
```

For another 10 byte register we are generating fault message:
```
lambda: |-
      std::string z = "";
      int idx = item->offset;
      //byte[0]
      if ((data[idx] & 0x1) != 0) z += "GridOVP,";
      if ((data[idx] & 0x2) != 0) z += "GridUVP,";
      if ((data[idx] & 0x4) != 0) z += "GridOFP,";
      if ((data[idx] & 0x8) != 0) z += "GridUFP,";
      if ((data[idx] & 0x10) != 0) z += "PVUVP,";
      if ((data[idx] & 0x20) != 0) z += "GridLVRT,";
      if ((data[idx] & 0x40) != 0) z += "reserve-ID7,";
      if ((data[idx] & 0x80) != 0) z += "reserve-ID8,";
      //byte[1]
      idx++;
      if ((data[idx] & 0x1) != 0) z += "PVOVP,";
      if ((data[idx] & 0x2) != 0) z += "IpvUnbalance,";
      if ((data[idx] & 0x4) != 0) z += "PvConfigSetWrong,";
      if ((data[idx] & 0x8) != 0) z += "GFCIFault,";
     .......any many more
```

Postprocessing scripts would be ideal for this. Especially since the entities indicated here are for diagnostics purpose, nobody will probably want to implement helpers to convert them into a useful format in HA. Specification should provide knowledge to decode this.

PR #65 (2024-12-13): value template, decimal places and fixes for mqtt discovery, contribution
Fixes:
 - Fixed issues with mqttvalue == 0 Closes #63
 - Added value template to entity Closes #36
     You can specify an optional value template to handle special values See home assistant documentation for information 
     about value templates

Features
 - Added Decimal places to numbers
 - Added no poll option to Poll Modes
closed 2024-12-13 by volkmarnissen