modbus2mqtt/modbus2mqtt #37: Unable to read coils on TCP

albgit, 2024-10-30:
I'm trying to add a coil to my working specification (with only holding registers), but it fails to read it.
I remeber that modbus2mqtt uses the same modbus api as nodered, so I tried to replicate the coil read there (coil 21).
Everything is working fine so I assume that is a modbus2mqtt problem.

Log:

-----------------------------------------------------------
 Add-on: Modbus <=> MQTT
 Service converts RTU and TCP Modbus to MQTT from other
-----------------------------------------------------------
 Add-on version: 0.16.11
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.10.2
 Home Assistant Supervisor: 2024.10.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
10/30/2024 9:19:01 PM notice HttpServerBase Hassio authentication successful
10/30/2024 9:19:01 PM notice modbus2mqtt modbus2mqtt listening on  37e92509-modbus2mqtt: 3000
10/30/2024 11:11:10 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 0 length:1  retrying ... 
10/30/2024 11:12:14 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 21 length:1  retrying ... 

nodered flow with coil reading:
[image]

modbus2mqtt, 2024-10-31:
I will look into it in two weeks.Von meinem iPhone gesendet

coil.nodered.png (view on web)

modbus2mqtt, 2024-10-31:
Can you please download the specification and attach it to the issue? (Menu/import/export) download link of the affected specification.I tested the coils with RTU only. May be that’s the reason.Von meinem iPhone gesendet

albgit, 2024-11-04:
I've attached the specification file, but I can't add the coil to it (the specification editor doesn't permit it).
This is the screenshoot of the editor:

[image]
[hmi-120.zip](https://github.com/user-attachments/files/17616373/hmi-120.zip)

volkmarnissen, 2024-11-11:
Thank you for attaching the specification.
It helped to get an idea of the issue.
However, I have no device with the same behaviour.
My devices have a single coil per address.
Your device seem to have 8 coils per address.

So, can you please help me?
Add three coils to your specification. Modbus Address = 1,2,3 (one by one)
Please let me know if they return values.

These are my expectations:

|Modbus2mqtt Address| Byte | Bit| description|
|-------|------- |-------|-------|
|1|1| 8 |Weekly Timer|
|2|2|16|Solar Heater|
|3 |3 |24 |Floor Debug|

Please let me know whether my expectations are correct.

You can also select the first bit of any other Byte to verify the value.

albgit, 2024-11-15:
If I understood it returns no results...

-----------------------------------------------------------
 Add-on: Modbus <=> MQTT
 Service converts RTU and TCP Modbus to MQTT from other
-----------------------------------------------------------
 Add-on version: 0.16.12
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2024.11.1
 Home Assistant Supervisor: 2024.11.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
11/15/2024 6:32:26 PM notice HttpServerBase Hassio authentication successful url:/api/hassio_ingress/V-WPNoBn6KfX9HNBTDRBbQ1DOannwEfEPmT8f8Ea7Dc
11/15/2024 6:32:26 PM notice modbus2mqtt modbus2mqtt listening on  37e92509-modbus2mqtt: 3000
11/15/2024 6:55:41 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 1 length:1  retrying ... 
11/15/2024 6:56:37 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 0 length:2  retrying ... 
11/15/2024 6:57:08 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 1 length:1  retrying ... 
11/15/2024 6:57:28 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 1 length:2  retrying ... 
11/15/2024 6:58:00 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 2 length:1  retrying ... 
11/15/2024 6:58:31 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 0 length:3  retrying ... 
11/15/2024 6:59:02 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 1 length:1  retrying ... 
11/15/2024 6:59:32 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 1 length:3  retrying ... 
11/15/2024 7:00:03 PM notice modbuscache http(3) readDiscreteInputs TIMEOUT: slave:3 address: 2 length:1  retrying ... 

[image]

arturmietek, 2024-11-19:
I also couldn't manage to read coils.
Here is screenshot from validation response:
[image]

And to be sure i also check reading coils from pymodbus using this piece of code:
`result  = client.read_input_registers(address=0x0000, count=0x01, slave=0x01)
print (result.registers)

result = client.read_coils(address=0x0000, count=0x01, slave=0x01)
print (result.bits)`

and got result:
`[3]
[True, False, False, False, False, False, False, False]`

Maybe there is something I don't understand here but I assume that first coil is active.

modbus2mqtt, 2024-11-19:
Yes, this is still not solved. I have no device with multiple coils per byte.Von meinem iPhone gesendet

image.png (view on web)

arturmietek, 2024-11-21:
In my case I would like to build driveway gate adapter. I started with Arduino UNO and UART to RS485 board. It is simple to make Modbus slave with https://github.com/CMB27/ModbusADU library. In this project i can use holding register to provide opening request but coils would be more elegant way.

Maybe i can make some debug for you?

albgit, 2024-11-27:
Can you explain: 
> If your device sticks to modbus standard with one coil (bit) per address, it works

?
I was thinking that my device was "standard" as in ha and node red I'm able to read every coil.
In my first example the coil "Quiet mode" has address "21" (as in specification and ha or node red configuration).
Reading the manufacturer specification there is a table (p. 46) that every bit can be adressed with the byte+bit, so for "Quiet mode" should be byte 3 bit 5.
Attached my specification doc.
[modbus-versati-iii-en.pdf](https://github.com/user-attachments/files/17935360/modbus-versati-iii-en.pdf)

modbus2mqtt, 2024-11-27:
You are right, your device should be a standard device.
However, I am not able to reproduce it with my devices, that's why I assumed, there is something different with the device.
My devices work properly with coils. So I don't know how to analyze it.
I'd like to add some more coils from my devices to a specification. May be then I am able to reproduce it.

albgit, 2024-11-28:
Maybe it can be helpful one of this simulators? They can simulate a Modbus TCP connection
[https://sourceforge.net/projects/pymodslave/](url) 
[https://unserver.xyz/products/modbus-slave-simulator/](url)

modbus2mqtt, 2024-11-28:
That's a good idea.

arturmietek, 2024-12-02:
Today i manage to start your code localy and find whats wrong with coils.
You read coils using readDiscreteInputs
`  modbus-serial {
  modbus-serial   action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 32 02 00 00 00 08 7c 0f>,
  modbus-serial   unitid: 50,
  modbus-serial   functionCode: 2,
  modbus-serial   length: 6
  modbus-serial } +1m
  modbus-serial {
  modbus-serial   action: 'receive serial rtu buffered port',
  modbus-serial   data: <Buffer 32 02 01 00 ae cc>,
  modbus-serial   buffer: <Buffer 32 02 01 00 ae cc>
  modbus-serial } +27ms
  modbus-serial {
  modbus-serial   action: 'emit data serial rtu buffered port',
  modbus-serial   buffer: <Buffer 32 02 01 00 ae cc>
  modbus-serial } +0ms`
https://www.modbustools.com/modbus.html#function02 
And that works for one of my devices that actually has discrete inputs with octocouplers to sens some external switches or input voltage but this device has 8 of them so i need to change length to 8. Otherwise its not working.

For coils we need function 0x01 and here is example of reading coils from same device with pymodbus:
DEBUG:pymodbus.logging:SEND: 0x32 0x1 0x0 0x0 0x0 0x8 0x38 0xf
DEBUG:pymodbus.logging:RECV: 0x32 0x1 0x1 0x3 0x1e 0xcd
It use function 0x01 and reads that coil 1 and 2 is on (0x3).

Maybe tomorrow i'll find more time to go further with this.

albgit, 2025-01-22:
Is there something I can do to help fix this bug?

arturmietek, 2025-01-22:
I made pull requests but We need to wait some time because volkmarnissen creating dev environment and He is on holiday so I hope fingers cross will be enough for this topic ;)

tomtom13, 2025-03-13:
Any update on this issue ? I'm trying to setup a coil on NETIO devices to control the device outputs.

modbus2mqtt, 2025-04-16:
Closed with Pull request 106
Please install modbus2mqtt.latest from [Modbus2mqttt Repository](https://github.com/modbus2mqtt/hassio-addon-repository).
It contains this pull request.
It's an early version. There might be a few improvements in following pull requests.

closed 2025-04-16 by modbus2mqtt