modbus2mqtt/modbus2mqtt #38: [BUG] High load on idle due to stale data, new container doesn't have the issue, even with the same spec / connections imported.

madejackson, 2024-10-31:
It seems like while I was trying to create a new spec I was able to invoke a high load from the container which I can't get rid of without erasing the data. Starting fresh and importing the spec from the previous data fixed the issue.

See comparison of old (modbus2mqtt) and new (modbus2mqtt2) container:
[image]

modbus2mqtt, 2024-10-31:
I assume the configuration of bus or slave leads to the issue.If possible, download the configuration and attach it to this issue. (Menu/ export specification)Von meinem iPhone gesendet

image.png (view on web)

madejackson, 2025-04-23:
no problem. I just uploaded the local folder from config.

closed 2025-09-03 by modbus2mqtt