modbus2mqtt/modbus2mqtt #39: [Manufacturer related] Unable to use negative temperatures

albgit, 2024-11-04:
In my specification there are several registers that have default negative values.
Adding it to modbus2mqtt gives me a positive integer.
In manufacturer pdf data types is "Unsigned Integer".
Example "Lower AT-Heat", value:-20℃, values by modbus2mqtt: 65516.
This is the specification:
[hmi-120.zip](https://github.com/user-attachments/files/17616507/hmi-120.zip)
Pdf from the manufacturer:
[modbus-versati-iii-en.pdf](https://github.com/user-attachments/files/17616519/modbus-versati-iii-en.pdf)

volkmarnissen, 2024-11-11:
This will be implemented in the next release.
Signed integer will be configurable in the number format list.

closed 2024-11-15 by volkmarnissen

albgit, 2024-11-15:
Updated to 0.16.12.
I changed the type to signed int16 with no changes (value 65516).

PR #45 (2024-11-16): Fix for signed int16 data, Restore public version in specifications, Update validation messages after modbus value change
- Fix for signed int16 data. Closes  #39
- Restore public version in specifications (Changes in specification and angular component)
- Update validation messages after modbus value change (angular)