modbus2mqtt/modbus2mqtt #40: Conflict with zigbee2mqtt - zigbee devices disappears

arturmietek, 2024-11-11:
After adding device or restart modbus2mqtt HA addon all of zigbee2mqtt devices disappears. I need to restart HA core and from this point zigbee and modbus devices are visible at the same time. 

BTW this project i awsome!!
I would like to contribute on SDM 620 Eastron energy counter, 2 sofarsolar inverters and one voltage and current accussition module when i finish preparing whole specification for them.

volkmarnissen, 2024-11-11:
This will be fixed with next release. (End of this month)
The zigbee2mqtt topic won't get lost, but you are right. they will be deleted in the current session.
As a workaround, restarting zigbee2mqtt addon should regenerate the entities.

If you want to copy eastron SDM 720d,:
Edit the SDM720d.
Change the specification file name and specification name.
This will create a local copy with the name new name, but all entities.

For contribution logon to your github account, create a Github application token,
goto the Addon, enter the github token in the configuration menu.

Then you can contribute your specification at menu/import/export specifications.
If all tests are successful, your specification will be published after a few minutes.

Usually I am busy with the project. So, each specification is really great help.

Just open issues, if you face something you need.
The project gets more and more stable, but there are still things which can be improved.

closed 2024-11-15 by volkmarnissen

arturmietek, 2024-12-23:
Tried to copy sdm720 specifiaction to make smd630-m and no luck because of:
`Error: Cannot rename a published file
    at ConfigSpecification.writeSpecificationFromFileSpec (/usr/app/node_modules/@modbus2mqtt/specification/dist/configspec.js:435:31)
    at ConfigSpecification.writeSpecification (/usr/app/node_modules/@modbus2mqtt/specification/dist/configspec.js:494:14)
    at /usr/app/node_modules/@modbus2mqtt/server/dist/httpserver.js:485:25
    at /usr/app/node_modules/@modbus2mqtt/server/dist/httpServerBase.js:98:13
    at Layer.handleRequest (/usr/app/node_modules/router/lib/layer.js:145:17)
    at next (/usr/app/node_modules/router/lib/route.js:159:13)
    at Route.dispatch (/usr/app/node_modules/router/lib/route.js:119:3)
    at handle (/usr/app/node_modules/router/index.js:428:11)
    at Layer.handleRequest (/usr/app/node_modules/router/lib/layer.js:145:17)
    at /usr/app/node_modules/router/index.js:293:15`

Also tried to upload zip file with renamed file names and specification name but after success massage there is now new speccification in the system avaliable.

modbus2mqtt, 2024-12-23:
You can open SDM720 specification and change the filename of the specification. This will create a copy of it.

arturmietek, 2024-12-23:
What do You mean "open specification"? I tried to change filename when editing specification associated to device and got error message from previous comment. Don't know how to open specification alone.

commit cf11a753 (2024-11-15): mqttdiscovery: avoid deletion of non modbus2mqtt discovery topics. closes #40
prettier