modbus2mqtt/modbus2mqtt #62: Set Unit of measurement for selected device classes

modbus2mqtt, 2024-11-26:
Some device_classes have a limited set of Unit of measurement.
If the user selects such a device class, the Unit of measurement field will be changed to select field and show only allowed value.
The first value in the list will be preselected.

arturmietek, 2024-11-28:
Today i tried to switch from HA internal modbus integration to modbus@mqtt for Eastron SDM120-M device.
Public specification dosen't define unit of measurment for total active energy so i tried to set it. After setting state class for one entity all of them lost theirs name for couple of seconds. Next names came back and device class, UOM, ans state class dissapears. Its not possible to set state class for me.

modbus2mqtt, 2024-11-29:
My experience with Unit of measurement are similar. I found out, that it's a good idea to set the device class first.
UOM is not always required. Sometimes the UOM won't. be recognized by Home Assistant.

One of the next updates of modbus2mqtt will set the list of allowed unit of measurements depending on the device class.
A few device classes have a restricted set of unit of measurements.
However, this won't solve your issue.

In general, I assume this is more a home assistant issue with MQTT Discovery.
Modbus2mqtt just sends device_class and unit_of_measurement to Home Assistant.
There is no own logic.
That's why I will close this issue.

This disappearing of the name is a new issue. I saw it already, but it should have no impact on the functionality.
Once I found out the reason, I will fix it. If you want to follow this up, please create a new issue.

arturmietek, 2024-11-30:
This is general problem with specification editor. Now i have separate HA with modbus@mqtt only (without mqtt).

When i set state class and click save fields "device class", "UOM", "state class" are cleared.
Here i example of api/modbus/specification request when i set only "device class", "UOM":
[image]
And here when i set all 3 fields:
[image]

BTW: i spent some time today to try setup project localy without luck so far :( Would be nice to see some instructions on this like you mention in documentation.

volkmarnissen, 2024-12-02:
Thank you for the analysis.
You are right. This will be fixed with the next version.

Documentation standalone:
There is already issue #25 which deals with standalone installation.
Currently, I am busy with fixing and improving the app, so this must still wait a bit.

PR #64 (2024-12-11): Hassio addon
Fixes:

 - Set Unit of measurement according to device class. Closes #62
 - Set Quality of service for a slave. Closes #61 
    - For command topics it's important to get the message. This can be achieved by setting QoS
      There is a QoS named Auto. It will set QOS to 1 if the slave has a writable entity. Otherwise it is 0 (sensors)
 - MQTT Discovery: 
     - Send MQTT discovery when changing the slave (or specification)
     - Send State update after receiving a command (topic)
 - Add debug information from MQTT Client  in log file using debug prefix "mqttclient"

Features:
   [... cut]
closed 2024-12-11 by volkmarnissen