modbus2mqtt/modbus2mqtt #67: Crash on new 0.16.16 version

albgit, 2024-12-15:
I've updated to 0.16.16 version and now after run modbus2mqtt crash.
I didn't change anything in the specification.
Here the log output:

> -----------------------------------------------------------
>  Add-on: Modbus <=> MQTT
>  Service converts RTU and TCP Modbus to MQTT from other
> -----------------------------------------------------------
>  Add-on version: 0.16.16
>  You are running the latest version of this add-on.
>  System: Home Assistant OS 14.0  (amd64 / generic-x86-64)
>  Home Assistant Core: 2024.12.2
>  Home Assistant Supervisor: 2024.11.4
> -----------------------------------------------------------
>  Please, share the above information when looking for help
>  or support in, e.g., GitHub, forums or the Discord chat.
> -----------------------------------------------------------
> s6-rc: info: service base-addon-banner successfully started
> s6-rc: info: service fix-attrs: starting
> s6-rc: info: service base-addon-log-level: starting
> s6-rc: info: service fix-attrs successfully started
> s6-rc: info: service base-addon-log-level successfully started
> s6-rc: info: service legacy-cont-init: starting
> s6-rc: info: service legacy-cont-init successfully started
> s6-rc: info: service legacy-services: starting
> s6-rc: info: service legacy-services successfully started
> 12/15/2024 4:54:00 PM notice modbus2mqtt angulardir is /usr/app/node_modules/@modbus2mqtt/angular/dist/browser
> From https://github.com/modbus2mqtt/modbus2mqtt.config
>    1cf079f..ba84e09  main       -> origin/main
> 12/15/2024 4:54:01 PM notice m2mGithub Updating 1cf079f..ba84e09
> 12/15/2024 4:54:01 PM notice m2mGithub Fast-forward
> 12/15/2024 4:54:01 PM notice m2mGithub  specifications/dimplexpco5.yaml     |  10 +-
> 12/15/2024 4:54:01 PM notice m2mGithub  specifications/eastronsdm720-m.yaml | 264 +++++++++++++++++++++---------------
> 12/15/2024 4:54:01 PM notice m2mGithub  2 files changed, 159 insertions(+), 115 deletions(-)
> 12/15/2024 4:54:01 PM notice m2mGithub 
> 12/15/2024 4:54:02 PM ERR! config Unable to call event listener: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:54:02 PM notice HttpServerBase Hassio authentication successful url:/api/hassio_ingress/V-WPNoBn6KfX9HNBTDRBbQ1DOannwEfEPmT8f8Ea7Dc
> 12/15/2024 4:54:02 PM notice modbus2mqtt modbus2mqtt listening on  37e92509-modbus2mqtt: 3000
> 12/15/2024 4:54:02 PM ERR! mqttdiscover poll: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:54:12 PM ERR! mqttdiscover poll: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:54:22 PM ERR! mqttdiscover poll: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:54:33 PM ERR! mqttdiscover poll: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:54:43 PM ERR! mqttdiscover poll: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:54:53 PM ERR! mqttdiscover poll: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:55:03 PM ERR! mqttdiscover poll: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:55:13 PM ERR! mqttdiscover poll: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:55:23 PM ERR! mqttdiscover poll: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:55:31 PM ERR! config Unable to call event listener: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> 12/15/2024 4:55:34 PM ERR! mqttdiscover poll: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
> node:internal/streams/writable:315
>       throw new ERR_INVALID_ARG_TYPE(
>       ^
> TypeError [ERR_INVALID_ARG_TYPE]: The "chunk" argument must be of type string or an instance of Buffer or Uint8Array. Received an instance of Array
>     at new NodeError (node:internal/errors:405:5)
>     at _write (node:internal/streams/writable:315:13)
>     at Writable.write (node:internal/streams/writable:337:10)
>     at writeStringOrBuffer (/usr/app/node_modules/mqtt-packet/writeToStream.js:889:12)
>     at connect (/usr/app/node_modules/mqtt-packet/writeToStream.js:269:5)
>     at Object.generate [as writeToStream] (/usr/app/node_modules/mqtt-packet/writeToStream.js:31:14)
>     at MqttClient._writePacket (/usr/app/node_modules/mqtt/build/lib/client.js:934:46)
>     at MqttClient.connect (/usr/app/node_modules/mqtt/build/lib/client.js:297:14)
>     at MqttClient._reconnect (/usr/app/node_modules/mqtt/build/lib/client.js:790:18)
>     at Timeout.<anonymous> (/usr/app/node_modules/mqtt/build/lib/client.js:806:22) {
>   code: 'ERR_INVALID_ARG_TYPE'
> }
> Node.js v18.20.3
> [16:55:36] INFO: Service  exited with code 1 (by signal 0)
> s6-rc: info: service legacy-services: stopping
> s6-rc: info: service legacy-services successfully stopped
> s6-rc: info: service legacy-cont-init: stopping
> s6-rc: info: service modbus2mqtt: stopping
> s6-rc: info: service modbus2mqtt successfully stopped
> s6-rc: info: service legacy-cont-init successfully stopped
> s6-rc: info: service fix-attrs: stopping
> s6-rc: info: service base-addon-log-level: stopping
> s6-rc: info: service fix-attrs successfully stopped
> s6-rc: info: service base-addon-log-level successfully stopped
> s6-rc: info: service base-addon-banner: stopping
> s6-rc: info: service base-addon-banner successfully stopped
> s6-rc: info: service s6rc-oneshot-runner: stopping
> s6-rc: info: service s6rc-oneshot-runner successfully stopped

albgit, 2024-12-15:
I found the same log from 0.16.15 version.
Last working version: 0.16.14

modbus2mqtt, 2024-12-15:
Is it possible to download and attach the specification?
This will help to reproduce the issue

modbus2mqtt, 2024-12-15:
Please attach the /data/local/modbus2mqtt.yaml as well.
If it contains a line "will: ... "
with an array as value, that could be a good explanation for the issue.

albgit, 2024-12-15:
Here the local zip with everything in it

[local 20241215.zip](https://github.com/user-attachments/files/18141634/local.20241215.zip)

modbus2mqtt, 2024-12-15:
Your/usr/share/hassio/addons/data/37e92509-modbus2mqtt/local/modbus2mqtt.yaml contains this lines:

will:
    topic: modbus2mqtt/will
    payload:
      - 71
      - 111
      - 111
      - 100
      - 98
      - 121
      - 101
      - 33
    qos: 1

If you remove the lines, the system should work as before.

However, I will update the addon. It will automatically remove then.
This will happen in the next release 0.16.17 
So, you can either wait for 0.16.17 or fix it beforehand.
0.16.17 should be released next week.

closed 2024-12-15 by modbus2mqtt

albgit, 2024-12-16:
As I can't modify addon internal files, I'll wait for the update.
Thanks