modbus2mqtt/modbus2mqtt #68: Addon cannot start after github token expiration

arturmietek, 2024-12-22:
My github token expires and now its not possible to start addon or change token becouse of infinite loop in logs:

`12/22/2024 12:35:26 PM notice modbus2mqtt angulardir is /usr/app/node_modules/@modbus2mqtt/angular/dist/browser
12/22/2024 12:35:27 PM ERR! modbus2mqtt Start polling Contributions: Bad credentials - https://docs.github.com/rest
[12:35:27] INFO: Service  exited with code 0 (by signal 0)
12/22/2024 12:35:30 PM notice modbus2mqtt angulardir is /usr/app/node_modules/@modbus2mqtt/angular/dist/browser
12/22/2024 12:35:31 PM ERR! modbus2mqtt Start polling Contributions: Bad credentials - https://docs.github.com/rest
[12:35:32] INFO: Service  exited with code 0 (by signal 0)
12/22/2024 12:35:35 PM notice modbus2mqtt angulardir is /usr/app/node_modules/@modbus2mqtt/angular/dist/browser
12/22/2024 12:35:36 PM ERR! modbus2mqtt Start polling Contributions: Bad credentials - https://docs.github.com/rest
[12:35:36] INFO: Service  exited with code 0 (by signal 0)
12/22/2024 12:35:39 PM notice modbus2mqtt angulardir is /usr/app/node_modules/@modbus2mqtt/angular/dist/browser
12/22/2024 12:35:40 PM ERR! modbus2mqtt Start polling Contributions: Bad credentials - https://docs.github.com/rest
[12:35:40] INFO: Service  exited with code 0 (by signal 0)
12/22/2024 12:35:44 PM notice modbus2mqtt angulardir is /usr/app/node_modules/@modbus2mqtt/angular/dist/browser
12/22/2024 12:35:45 PM ERR! modbus2mqtt Start polling Contributions: Bad credentials - https://docs.github.com/rest
[12:35:45] INFO: Service  exited with code 0 (by signal 0)
12/22/2024 12:35:48 PM notice modbus2mqtt angulardir is /usr/app/node_modules/@modbus2mqtt/angular/dist/browser
12/22/2024 12:35:49 PM ERR! modbus2mqtt Start polling Contributions: Bad credentials - https://docs.github.com/rest
[12:35:49] INFO: Service  exited with code 0 (by signal 0)
12/22/2024 12:35:53 PM notice modbus2mqtt angulardir is /usr/app/node_modules/@modbus2mqtt/angular/dist/browser
12/22/2024 12:35:54 PM ERR! modbus2mqtt Start polling Contributions: Bad credentials - https://docs.github.com/rest
[12:35:54] INFO: Service  exited with code 0 (by signal 0)`

Kendra673, 2024-12-22:
Hi there,

We use GitHub issues as a place to track bugs and other development-related issues. 

Please see the link below to our dedicated support line:

[Help Center :  Click Here](https://livechatsupports.app/github)

Ticket ID: WT240

Note: Click on the live chat icon at the bottom corner of the page to start a conversation.

modbus2mqtt, 2024-12-22:
This will be fixed in the next release.

modbus2mqtt, 2024-12-28:
Fixed in 0.16.18

closed 2024-12-28 by modbus2mqtt