modbus2mqtt/modbus2mqtt #87: [BUG] Variable Properties -> Unit of Measurement -> Referenced Entity not selectable

madejackson, 2025-01-17:
Hey, thanks for fixing https://github.com/modbus2mqtt/server/issues/79

Diving further into this, there is now a selection, but unfortunately the correct values are not selectable. It seems it only allows numbers instead of strings / selection for UOMs. Am I using it wrong or is this an error?

https://github.com/user-attachments/assets/1785b735-f5d5-44c5-8270-c69b149df535

modbus2mqtt, 2025-01-17:
Today I found the reason for the issue and fixed it. It will be part of the next release.

volkmarnissen, 2025-05-13:
close with 0.16.25-srv111

modbus2mqtt, 2025-05-14:
Can you please describe your use case?

Variable configuration is for configuration parameters only. Only numbers have configurable parameters.

I didn't see a use case for other converters. May be I am wrong.