modbus2mqtt/modbus2mqtt #89: Error on starting the add-on

romromp, 2025-01-19:
Hello,

I'm trying to install this add on on HASSIO. 

I Have this error 

TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received undefined
    at new NodeError (node:internal/errors:405:5)
    at validateString (node:internal/validators:162:11)
    at join (node:path:1171:7)
    at HttpServer.sendIndexFile (/usr/app/node_modules/@modbus2mqtt/server/dist/httpServerBase.js:199:36)
    at HttpServer.processAll (/usr/app/node_modules/@modbus2mqtt/server/dist/httpServerBase.js:246:14)
    at Layer.handleRequest (/usr/app/node_modules/router/lib/layer.js:145:17)
    at next (/usr/app/node_modules/router/lib/route.js:159:13)
    at Route.dispatch (/usr/app/node_modules/router/lib/route.js:119:3)
    at handle (/usr/app/node_modules/router/index.js:428:11)
    at Layer.handleRequest (/usr/app/node_modules/router/lib/layer.js:145:17)

modbus2mqtt, 2025-01-20:
Which version of the addon are you using?

cubitusgp, 2025-02-09:
I think Add-on version: 0.16.25 I have same error with HA last version

cubitusgp, 2025-02-10:
This error is loged when I click on open Webui. service is listening but there is an error on webpage.

modbus2mqtt, 2025-02-10:
Hi,

currently, I have no explanation for this. Is it possible for you to use0.16.23?
Currently there are a few changes ongoing may be they made it to the release. 
In
March there will be a new release which should solve it.

cubitusgp, 2025-02-10:
Hi dear,

On home assistant, I can't use an older version, I try to do it with a ssh 22222 command line. I installed another instance on portainer.

cubitusgp, 2025-02-10:
Can you create a repository for 0.16.23 version ?

I tried to modify config.yaml file but HA OS still install 0.16.25 version

volkmarnissen, 2025-05-13:
Please try 0.16.25-srv111 frpm modbus2mqtt.latest

closed 2025-05-13 by volkmarnissen