modbus2mqtt/modbus2mqtt #91: Add possibility to convert the byte and the word order for numbers and also for texts

lsimma, 2025-01-20:
Hello
Can you add the possibility to convert the byte and the word order for numbers and also for texts, please?
Device is named "Pro3EM-1" but shown as "rP3oME1-".
If there is another possibility to to this please point me to the information, please.

[image]

[image]

Best regards,
Lukas

modbus2mqtt, 2025-01-20:
Hi,

thank you for this feedback. Yes, it's a good idea to add support for different byte order.
However, I don't have a device for testing, but you can help me:

I need test data. You can downloadyour specification  it from Menu / Contribute/import/export Specification.
Just click on the download link and send me the downloaded file attached to this issue.
If possible, it would be great to get the expected values also.

Additionally question: Are the values correct for unsigned 16 bit?

The fix be implemented by end of next month or begin of march.

lsimma, 2025-02-02:
Hello

Tested float32 with voltage, the value is 230V
Modbust register:
  * 1020 and 1021
Result:
* If the higher Modbus Address 1021 is used in the Modbus2MQT definition for float32 then the value is correct

------------------ DETAILS -----------------
Shelly definition for Voltage:
Modbus Register Address is 1020 and 1021)

[image]

Modbus2MQTT definition to "debug" the two register addresses with uint16:
[image]

Modbus2MQTT definition for float32, on is defined with register address 1020, the other with 1021
[image]

MQTT results:
[image]

[image]

Calculation to check it shown value is a valid and correct float32 number
[image]

[image]

[image]

lsimma, 2025-02-02:
read unsigned 32 bit interger 
INFO: I had to select "signed int32" to geht an unsigned result.

a) This test show that unsinged int32 read the addresses in a different sequence in compare to float32
* unsingned int 32: if register 1000 was defined then reg1000 and reg1001 was read
* float32: if register 1021 was defined then reg1020 and reg1021 was read

b) This test show that the resulting value for unsigned int32 use different word order then float32

[image]

lsimma, 2025-02-03:
Testdata for above described float32 and uint32 issues:

[shellypro3em test 2025-02-03.zip](https://github.com/user-attachments/files/18634777/shellypro3em.test.2025-02-03.zip)

arturmietek, 2025-02-03:
Do I understand correctly that we need support for both big endian and little endian?
If that correct It's should be configured in number conversion properties near number format.
All devices I checked so far uses BE.

modbus2mqtt, 2025-02-04:
If I am not wrong, Little endian/Big endian is a feature of the specification. It's not necessary to configure it for every entity individually.
E.g Shelly devices have always the same endian type.

modbus2mqtt, 2025-02-04:
I was assuming BE is enough for most of the cases. This seems to be wrong. So, yes LE needs also support.

volkmarnissen, 2025-02-07:
Modbus transfers always 16-Bit unsigned Big endian. (Specification)
Conversion to 32bit, float32… is not directly supported. So, there can be more implementations to Convert Numbers and Texts.
The Output is always 64-Bit float.
For 32-Bit signed or unsigned, it‘s important to know if the high Word or low Word is First. It has nothing to do with endian. Because  the Words are already available in the correct byte order.
For float Input it‘s different. If the slave is little endian, the big endian words must be swapped and then converted via readfloatLE.
If it’s Big Endian, no conversion is needed, but the conversion must be done with readfloatBE.
For text, only the word might need to be swapped. This is not endian specific, but specific to the implementation.

Currently, the modbus values are read into number array. The bytes are extracted from that array. I assume, this implementation needs to be reworked.

So, there are the following conversion options
For Text: swap bytes ? Boolean
For 32-bit un-/signed: low word first? Boolean
For float 32: swap bytes?
BE/LE . I am not sure whether swap bytes is required. I assume, if float is LE byte swap is required. Otherwise it is not required.

In this case, it makes more sense to add
un-/signed32 swapped and
float LE
floatBE
to the existing data formats
The text properties need a swap byte check box

closed 2025-10-02 by volkmarnissen