modbus2mqtt/modbus2mqtt #100: Add-on "removed from repository" notification

albgit, 2025-03-04:
Hi,
On home assistant I've this notification: 

> Add-on Modbus <=> MQTT has been removed from the repository it was installed from. This means it will not get updates, and backups may not be restored correctly as the supervisor may not be able to build/download the resources required.
> Selecting Submit will uninstall this deprecated add-on. Alternatively, you can check Home Assistant help (https://www.home-assistant.io/help/) and the community forum (https://community.home-assistant.io/) for alternatives to migrate to.

I don't understand what this problem refers to as everything is working as before this notification.

volkmarnissen, 2025-03-04:
This is a temporary issue, it will be resolved soon.

modbus2mqtt, 2025-03-14:
It's an issue with github/homeassistant and this addon.

The released versions of the addon are in the release branch.
They were in the main branch before.
Github is configured to use the  release branch as default.
So, if you open https://github.com/modbus2mqtt/hassio-addon-repository in web browser, you'll see the release branch.
Homeassistant will still show the main branch (which is a not working dev version)

However, there is a workaround:
Use the release branch of the addon repository:
[https://github.com/modbus2mqtt/hassio-addon-repository#release](https://github.com/modbus2mqtt/hassio-addon-repository#release)

You need to uninstall the addon in homeassistant, then remove the repository from addon store,
add the release repository link again.
https://github.com/modbus2mqtt/hassio-addon-repository#release

modbus2mqtt, 2025-03-14:
This is solved now. No need for the work around

closed 2025-03-14 by modbus2mqtt