modbus2mqtt/modbus2mqtt #107: Timeout after docker restart

madejackson, 2025-04-16:
Hey,

so just pulled the v26 trying to setup a new template for my devices.

First, after a restart, modbus2mqtt is somehow unable to get new values from modbus. The TCP-Server is working flawlessly as I can read those values without issues from modbus poll.

It seems to be happening with more than 5-6 Registers, below it may suddenly start working. Timeout is at 1000ms as test.
4/16/2025 4:52:09 PM notice modbuscache poll(1) readCoils TIMEOUT: slave:1 address: 0 length:10  retrying ... 

[image]

Not sure what else I can give you as infos.

modbus2mqtt, 2025-04-16:
Please pull from this new source:
```
docker pull ghcr.io/modbus2mqtt/modbus2mqtt.latest:0.16.26-srv106
```
It contains the new release with the latest pull requests.

You can as well install the homeassistant addon modbus2mqtt.latest

The new version has a refactored modbus handling. It might not be perfect now, but it has a lot more potential than the previous one. So, the old one **modbuscache** won't be supported in the future.

closed 2025-04-16 by modbus2mqtt

madejackson, 2025-04-23:
thx. Just commenting to confirm:
`ghcr.io/modbus2mqtt/modbus2mqtt.latest:0.16.26-srv106` works much better than `ghcr.io/modbus2mqtt/modbus2mqtt:latest`