modbus2mqtt/modbus2mqtt #109: [BUG] Modbus-Connection fails on high input load

madejackson, 2025-04-23:
Hey thanks for the continuous updates and improvements.

with `.latest:0.16.26-srv106` I'm able to setup far more to my template than I was able before. 

Right now I have 24x DO and 32x DI in 1x Slave (connected internally to IO-Controller via USB/UART to virtual serial port) The speed is far above 115K baud.

Unfortunately, I'm able to kill modbus2mqtt fairly easy on somewhat higher "load" Here an example where it fails to update some of the IO's until it freezes completely, as soon as I toggle it faster (default interval 1000ms, timeout 500ms):

https://github.com/user-attachments/assets/e0926c72-a9e1-4fb5-b9d7-730c1a8ce719

[image]

volkmarnissen, 2025-05-08:
Hey,

thank you again for your contructive feedback.
The latest changes were due to a refactoring of the modbus access based on your previous issue.
I am happy they helped you at least a bit.

I don't understand your scenario completely.
Let me try to explain what I understood:
- Your device is RTU based. It's connected with a high baud rate.
- Your device is connected to modbus2mqtt via TCP-> RTU bridge.
- You are triggering a button in homeassistant.

This works as long as you toggle (very?) slowly.
If you toggle quickly, modbus2mqtt fails.

Did I understand correctly?

Some comments:
- There is already a queue for the modbus requests (after refactoring).
- Due to limitations of the modbus api, coils must be processed one by one.
- Additionally in case of crc errors a restart of the connection is required.
- triggering a button in homeassistant sends a request immidiatetly to modbus2mqtt.
- reading of the current state happens on poll interval speed I hope this meets your requirements.
- Currently, there is one Queue Processor only. It's for RTU (and TCP). So, restarts may be required.
  I'd like to optimize this processor first. This will make the RTU access more stable.
- Later I'll add a TCP Queue Processor. It will have no error handling.
  This will prevent reconnections.

As of now, I don't know what kind of modbus error is causing the restart. A timeout should not reconnect.
Is it possible for you to reproduce the scenario and send the complete log of modbus2mqtt?

Please wait for the 0.16.25-srv111. I'll be available either today or tomorrow and contains a few fixes for error handling.
server. I won't promise, that it already solves your issue, but may be it does.

modbus2mqtt has also a REST API. May be this is beneficial to you. It does no polling, but reads on request.
You find the URL's of the REST api in the slaves view.

volkmarnissen, 2025-05-13:
0.16.25-srv111 is available now. I assume, it will solve your issue at least partly.

There were some performance eaters in previous versions.

madejackson, 2025-05-14:
> 0.16.25-srv111 is available now. I assume, it will solve your issue at least partly.
> 
> There were some performance eaters in previous versions.

Can't find that version, but I'm trying `.latest:0.16.26-srv112` now.

volkmarnissen, 2025-05-14:
That's the right one. 113 is also fine. It contains two small fixes most likely not relevant for your scenario.

I am not planning further refactorings. So, the next pull requests will be fixes only.

modbus2mqtt, 2025-05-15:
0.16.26-srv115 was just created. It contains important fixes for you.
There is a good chance for solving this issue.
Please let me know how it works.

madejackson, 2025-05-23:
Hey, I just tested it with 0.16.26-srv120 and it seems to work flawlessly now. The isue above cannot be reproduced so I'm gonna close this :) Thanks.

closed 2025-05-23 by madejackson